rotorflight/rotorflight-lua-ethos-suite #308: "Arming Disable Flags"  telemetry sensor name too long - truncated as "Arming Disable "

bob01, 2025-03-06:
Ethos telemetry source name seems to be limited to 15 chars ie "Arming Disable Flags" is truncated to "Arming Disable "

Test
system.getSource("Arming Disable Flags") -> nil
system.getSource("Arming Disable ") -> expected tele source

If you look carefully at the sensor name in the list on the ethos [Model] -> [Telemetry] page you can just see the trailing space

Using Ethos 1.6.2

robthomson, 2025-03-06:
I have confirmed with bertrand that this is 15 on all radios.

So we need to adjust

robthomson, 2025-03-06:
We need to cross check here:

https://github.com/rotorflight/rotorflight-lua-ethos-suite/blob/master/scripts/rfsuite/tasks/sensors/elrs.lua

https://github.com/rotorflight/rotorflight-lua-ethos-suite/blob/master/scripts/rfsuite/tasks/sensors/frsky.lua

https://github.com/rotorflight/rotorflight-lua-ethos-suite/blob/master/scripts/rfsuite/tasks/sensors/frsky_legacy.lua

Might be worth a global file with the names in that is used by all 3 scripts.

closed 2025-03-06 by robthomson

bob01, 2025-03-06:
👍