rotorflight/rotorflight-lua-ethos-suite #328: Rf Status acquiring voltage info on powerup

jimmy6616, 2025-03-09:
The status values from Telemetry take approx 25seconds to fully update the accurate values after power-up, looking at the poll rate on certain sensors they only update every 3.5 to 6seconds

robthomson, 2025-03-09:
That, seems an awfully long time.

Are these specifically esc sensors?

I am wondering if this is actually a delay on acquiring them on the fbl end - and not on the rotorflight end?

robthomson, 2025-03-10:
Thing is msp is not related to the telemetry at all.

The widget simply displays the sensor values once the arrive, whereas the msp stuff only happens when you load the system tool.

jimmy6616, 2025-03-10:
Just tested it and it worked immediately, let’s close this issue Rob
J

closed 2025-03-10 by robthomson