rotorflight/rotorflight-lua-ethos-suite #453: Documentation

jasc76, 2025-03-19:
Hi

I really struggled with the documentation regarding Ethos. I wanted to redo some stuff and really had to dig and had some head scratchers even though I already use the scripts since their release....
- The difference between touch and text version is not in the documentation. At first I always ended up with the text version (https://www.rotorflight.org/docs/next/setup/lua-scripts). Somewhere I stumbled across the mentioning of the touch script until I finally found it
- The status widget is nowhere mentioned and explained. This is sad, because new users might not know

My suggestions are:

- For new users it should be mentioned that there are two scripts, one for settings and one for the status
- explain that there is a text and a touch version
- Maybe the repos can be differentiated between touch and text, or the releases renamed. 
- The descriptions of the releases are 99% RF related. Maybe this can be reduced to the things regarding the LUA release and the changes within the release

With the Vantac RF controllers I'm sure there will be a push in the Frsky ecosystem and your scripts are **such a great great thing**, nobody should struggle getting and using them. Without them its a DIY project. Your scripts make it a real solution !

jimmy6616, 2025-03-19:
Click the Ethos button

[image]

jimmy6616, 2025-03-19:
and the status lua

https://www.rotorflight.org/docs/next/setup/radio-setup/radio-setup-ethos/ethos-nexus-status

jasc76, 2025-03-19:
Well, ok....
I didnt check the radio setup stuff; just the LUA section, because I wanted to install the LUAs ....

robthomson, 2025-03-19:
We are working hard on getting more docs.  Its taking time; and always ends up being one of the last things done.

If you fancy helping with documentation - would be great.  We are very under resourced and could do with help on it.

Closing this now as its not really an issue per say... if you are interested in helping - please touch base via discord.

closed 2025-03-19 by robthomson

jasc76, 2025-03-20:
I looked into the documentation stuff. And you use this generator and I don't have to time to get into that, because there was something with stuff to install and so on. I can, however, see to provide some content. Which discord?

robthomson, 2025-03-20:
Here you go..

https://discord.com/invite/f7fQyZY6