rotorflight/rotorflight-lua-ethos-suite #663: Dashboard X14 format

jimmy6616, 2025-05-25:
In the default inflight screen the text formatting is skew. Attached 

[image]

AERCHeli, 2025-06-11:
Hey @jimmy6616 , the dashboards have had a massive overhaul recently. You may want to consider downloading the latest master branch as I suspect it will resolve this issue. Keep in mind we are still very actively working on dashboards daily, so it may also pay to sit tight until we have everything finalised which shouldnt be too far away.

jimmy6616, 2025-06-11:
Thanks for your help, will do

AERCHeli, 2025-06-17:
Closing this issue down, tested on my own X14S and all appears as it should now. Any further issues, feel free to re-open.

closed 2025-06-17 by AERCHeli