rotorflight/rotorflight-lua-ethos-suite #757: Ethos logging error

rellik75, 2025-06-03:
Ethos 1.6.2 consistently throws error. See attached. 

[image]

robthomson, 2025-06-03:
That's indeed curious as not happening on our test boxes.

Are you using sd or internal storage.

rellik75, 2025-06-03:
Version 1.6.2 of Ethos. It’s an x20 pro so I’m saving to flash. And I deleted all ethos logs to make sure it wasn’t a space issue.

robthomson, 2025-06-03:
Very odd.

I also have the prodo you have a logs folder on your sdcard root?

robthomson, 2025-06-03:
I suspect a chkdsk may be needed.

rellik75, 2025-06-03:
I don’t use the SD card for logging so no logs folder exists. Nothing has changed on my radio between the last version of rotorflight and installing the latest release and I never got this error previously.

robthomson, 2025-06-22:
This is resolved in latest masters

closed 2025-06-22 by robthomson