rotorflight/rotorflight-lua-ethos-suite #983: Dashboard fuel gauge not showing expected value

jimmy6616, 2025-06-30:
Suite battery settings are set with a Consumption warning of 20%. X14 with a battery connected at 23.54V which equates to about 66%. The smartfuel gauge is showing 26%. Using the latest master as of the date\time of this post

AERCHeli, 2025-07-01:
Hey @jimmy6616 ,

I need some more info, is this a fully charged battery that you have connected and then flown down to 23.54v / 3.92v per cell? Or did you connect a battery to the heli that was already discharged down to 23.54v / 3.92v per cell?

If the scenario is that you connected a battery that was already discharged down to 23.54v / 3.92v per cell - The value you have is expected behavior with the new smart fuel logic. Unfortunately Lipo battery discharging is not linear, so determining usable capacity of a battery based on voltage is difficult, which is what the smart fuel sensor does if the charge level of the battery is below the full cell voltage level configured in RotorFlight. Whilst we could introduce far more complicated math that accounts for the discharge of a lipo not being linear, it introduces risks of miscalculated fuel% levels when a battery that is connected is partially discharged which could lead to the pilot over-discharging their pack - so we went for the 'safe' and conservative option and set it up so that in this scenario, smart fuel will return a very conservative fuel level to fly safely down to 0% to ensure you don't over-discharge your pack (last thing we want is to return values that are too high which could lead to people ruining packs and/or crashing).

If you didnt and it was a full pack and you flew it down to this level, I will look into it further as I have not seen this issue myself and it could be something to do with your consumption figures from your ESC being wrong, in which case you may need to look at tweaking consumption correction within RotorFlight.

AERCHeli, 2025-07-01:
Hey Jimmy,

As I mentioned in my previous comment, that would be in an ideal world where lipo battery discharging is linear, there are so many variables that come into play when calculating the remaining % of a battery based on voltage (ESC telemetry accuracy, lipo age, lipo chemistry, battery values being configured correctly in RotorFlight etc etc). We took the safe approach to calculating this so we under-estimate what is left in this scenario (it is VERY accurate if you plug in a battery that is above the full cell voltage value). 

I will take another look at the math and see if we can perhaps not be as safe as we are in the calculation to bring it closer BUT as I mentioned before, there is a balance that needs to be struck here between what's safe and what could lead to issues and at the end of the day, we wanted to ensure this is as safe as possible. The reality is, with all of the telemetry we have available - you can make your own judgement as to how long you can fly the pack based on the information you have. You can also make this return a higher fuel% value by lowering the min cell voltage value in battery settings, as this will widen the calculation that is used and give you a larger remaining % - although personally i'd strongly suggest to just fly to 0% from the value that is returned - its the safest option to avoid any issues.

jimmy6616, 2025-07-02:
Thanks Andrew let’s close the call
Jim

closed 2025-07-02 by AERCHeli