rotorflight/rotorflight-lua-ethos-suite #1086: Smart Fuel wrong percentage

vhuzalo, 2025-07-16:
I noticed that smart fuel sensor does not calculate the remaining percentage with precision. 

In this case i do not have current sensor, just voltage from esc telemetry

For example

Dashboard shows 52% 

[image]

Battery meter estimates 86%
[image]

These are battery sensors used:
[image]

robthomson, 2025-07-16:
There are a few things to take note of.. And I am sure @AERCHeli will comment here too.

1.  Without a current sensor this will currently not work too well
2. Fuel is a capacity remaining until 0%.  But 0% is more like 35% as this is when you should land. -so scaled to land and 0%

vhuzalo, 2025-07-16:
The main idea comes from this widget:

https://github.com/PD5DJ/AdvGauge/blob/main/advgauge/main.lua

vhuzalo, 2025-07-16:
Maybe I can help with this :)

robthomson, 2025-07-16:
Always happy to have extra help.

That's being said I know @AERCHeli was planning to look at it this week.

I don't mind if you want to look at it, but would prob need to run your through ALLOT of stuff first as the way the system works does require a bit of getting familiar

Message me on discord.

AERCHeli, 2025-07-17:
Happy to have you jump in and help with this one, I was planning on doing a simple lookup table to return more accurate percentages based on voltage that has a simple table for standard and HV lipo's based on discharge curves. The current way its calculated was intentionally done in a simple way that is 'safe', but happy to get this more accurate. 

We also need to look at tracking fuel% based on voltage when a consumption sensor isnt available, I spoke to Rob earlier, happy for you to jump on in and have a go at sorting it!